mikaelpatel/Arduino-Scheduler #2: Add support for SAMD based board

mikaelpatel, 2016-02-06:
Testers required.

mikaelpatel, 2016-02-06:
Please see commit https://github.com/mikaelpatel/Arduino-Scheduler/commit/90d94412d0d4009b8c8cc063cb47f2bf7e4073aa.

malachib, 2016-02-07:
Tests performed on Sparkfun SAMD21G mini breakout:

**SchedulerBenchmark: started**
1. Initiate scheduler and main task: 2 us
2. Yield main task: 2.60 us
3. Start a task: 6 us
4. Yield between two tasks: 5.65 us
5. Delay 10 ms and check increments: 9128:1608, 5.68 us
6. Start 25 tasks: 96:25, 3.84 us
7. Yield and check increments: 74:26, 2.85 us
8. Delay 10 ms and check increments: 9199:2912, 3.16 us

**SchedulerBenchmarkQueue: started**
3950:loop::count=101188, us=9.88
4950:loop::count=101152, us=9.89
5950:loop::count=101147, us=9.89
6950:loop::count=101152, us=9.89
7950:loop::count=101153, us=9.89
8950:loop::count=101145, us=9.89
9950:loop::count=101146, us=9.89
10950:loop::count=101157, us=9.89
11950:loop::count=101153, us=9.89
12950:loop::count=101151, us=9.89
13950:loop::count=101153, us=9.89
14950:loop::count=101146, us=9.89
15950:loop::count=101151, us=9.89
16950:loop::count=101147, us=9.89
17950:loop::count=101150, us=9.89
18950:loop::count=101145, us=9.89

Also tested "SchedulerEvent" example and seems OK.

Note I had to modify examples to use SerialUSB instead of Serial, as is typical for SAMD boards according to what I read.  If you like I can generate a pull request with the mods for those 3 examples

mikaelpatel, 2016-02-07:
Thanks! Updated the README with the results.

closed 2016-02-14 by mikaelpatel