mikaelpatel/Arduino-Scheduler #6: Enhance: Support ESP8266

mikaelpatel, 2016-03-26:
This could be done by using the continuation stack support (cont.h https://github.com/esp8266/Arduino/blob/master/cores/esp8266/cont.h).

soubok, 2016-08-21:
By adding some missing defines in scheduler.h, I can use the lib on my ESP-12:

```
#elif defined(ARDUINO_ARCH_ESP8266)

      /** Default stack size and stack max. */
      static const size_t DEFAULT_STACK_SIZE = 512;

      // https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map
      #define RAMSTART     (0x3FFE8000)
      #define RAMSIZE      (0x14000)
      #define RAMEND       (RAMSTART + RAMSIZE - 1)

```

mikaelpatel, 2016-08-21:
@soubok Thanks for the update. That was a neat solution. Have you run any of the examples or benchmarks? Cheers!

soubok, 2016-08-21:
@mikaelpatel, at the moment I just tried to run 2 concurrent tasks with some serial output, nothing more.

mikaelpatel, 2016-08-25:
@soubok Have you been able to run any of the examples and/or got your test with 2 concurrent tasks working? I am consider updating the library with the above definition but do not have any hardware to test this on.

anmaped, 2016-10-19:
@mikaelpatel I will test it on my esp8266 project with several continuations. Arduino for esp8266 does not manage continuations using a so elegant way.

anmaped, 2016-10-19:
@mikaelpatel Integration with continuation is not done. The proposal of the member @soubok is totally incomplete.

mikaelpatel, 2016-10-19:
@anmaped Thanks for your efforted testing this library for the ESP. I have done an update but not come around to pushing it mainly due to lack of testing. If I understand you correctly the suggested update (together with update of the ifdef in the source code) works. What default stack size would you suggest? 1K?

anmaped, 2016-10-19:
@mikaelpatel I will do a pull request with my changes. What examples do you want to test first ?

mikaelpatel, 2016-10-19:
@anmaped Would appreciate if you could run and verify all the example sketches before the pull request. I do not have any ESP boards at this time.

anmaped, 2016-10-19:
@mikaelpatel Okay. I will do that. Before I need to ask Arduino esp8266 to weak three functions, `esp_schedule`, ` esp_yield`, `optimistic_yield`, and finally `loop_task`.

anmaped, 2016-10-20:
Some preliminary results.

```
2. Yield main task: 1.04 us
3. Start a task: 1 us
4. Yield between two tasks: 1.66 us
5. Delay 10 ms and check increments: 0:0, nan us
6. Start 1 tasks: 6:1, 6.00 us
7. Yield and check increments: 3:2, 1.50 us
8. Delay 10 ms and check increments: 0:0, nan us
```

Delay is not yet implemented. There is room for one tasks only with standard Arduino esp8266 settings. The initiate scheduler cannot be called since is already initialized.

closed 2017-02-06 by mikaelpatel