mikaelpatel/Arduino-Scheduler #8: Documentation

leseigneur, 2016-05-16:
Hello.
I find your code and examples very valuable. I am looking for a documentation, have you wrote any you are willing to share ?
Or maybe the best way is to read all .h code and try all examples...
Anyway, great contribution, thanks.

mikaelpatel, 2016-05-16:
Thanks for your interest in this very small library for the Arduino. 

An API documentation file (HTML) can be generated from the source code with Doxygen but as this API is very small, basically only one member function, the header file is good reading. 

My guess is that you are looking for documentation on how to apply this to the Arduino core? That is much more documentation and needs a lot of explaining of issues when using concurrent programming mechanisms. The example set may give a good start. 

I only have additional teaching material (slides etc). Unfortunately is part of my consulting so it is not open-source at this time.

leseigneur, 2016-05-23:
Hello,
Thank you for your response.
In fact, my question does not refer to Arduino core but to a programming 
problem I encounter with this library. How do you manage to handle a 
queue without non-blocking queue reading ?

If a loop needs to react both to time event (delay) and to queue events, 
I need to be able to check queue state in a non blocking way.
Either I modify your library, either there is a way with the current 
library I may have overlooked.
Regards
Jean-yves

leseigneur, 2016-05-23:
Ok, sorry, I have just seen the available() method, it makes my day :)

closed 2016-05-28 by mikaelpatel