mikaelpatel/Arduino-Scheduler #9: remaining stack

soubok, 2016-06-14:
Hello,
I encounter some stack overflow issues and I wondering how to determine the remaining stack for the current thread.
Thanks.

mikaelpatel, 2016-06-14:
The member function stack() returns the approximate consumed stack in bytes. This can be called in the deepest function call tree (call frames and local variables). The thread stack must be able hold the deepest function call and any ISR with necessären stack space (min 32 bytes).

soubok, 2016-06-15:
Thanks.
this mean that thread's stackSize - Scheduler::stack() is the remaining available stack space.
I would like to probe the used stack of the current thread using something like that:

  int size = stackSize - Scheduler::stack();
  char ptr[size]; // or alloca()
  memset(ptr, 'X', size);

then test, a some points of my program, the how far the 'X' are overwritten.
Do you think it's doable ?

mikaelpatel, 2016-06-15:
@soubok Yes, that is possible. It is a traditional method of stack checking. It does not solve the overrun issue but allows trimming the stack size. The memset() could be done in https://github.com/mikaelpatel/Arduino-Scheduler/blob/master/Scheduler.cpp#L127. There is also need for a max_stack() member function to probe the maximum stack size during the test run.

closed 2016-12-04 by mikaelpatel