mikaelpatel/Arduino-Scheduler #13: improvement request: enable tasks to be terminated, e.g., by a  Scheduler.stopLoop() method ?

shiftleftplusone, 2017-01-25:
hello,
would you perhaps consider also to enable tasks to be terminated (e.g., by local expressions or even by environmental semaphores)?
e.g., by a  Scheduler.stopLoop() method?

Then it would be possible not only to start perpetual tasks but also some which are running just temporarily and optionally may be restarted later again in case of need.

Example:
Creating PID tasks for different motors, running independently and simultaneously until the targeted end condition has been reached, and then terminate, and afterwards can be re-started for a new PID target.

I just created such a thing using gcc feat. pthread on my Raspberry Pi.

mikaelpatel, 2017-01-25:
@shiftleftplusone A quick question: Where would the taskname come from? Seems like another API change or alternative method of starting a task. Hard to stop something by-name when it is never given a name.

shiftleftplusone, 2017-01-25:
the taskname would be the name which is also used in setup for Scheduler.startLoop(taskname), e.g. sth like
void loop2() {...}
void startblink() {...}
void getkbdstroke()  {...}

mikaelpatel, 2017-01-25:
In general this is not easy to do as the tasks are allocated on the main stack. The behavior needs to be specified. What does running and stopLoop mean? There could be several tasks running the same loop function. Is running that there is at least one task with that loop function? Implementing stopLoop as removing the task from the task queue is often not a good thing to do as the task May have to release resources. Cancel in pthreads is a bit more complex.

shiftleftplusone, 2017-01-25:
I may perhaps compare it with POSIX pthread because I am quite familiar to that 
- and I guess you are as well:
```
pthread_t   thread1ID,  thread2ID,  thread2ID;
   
    pthread_create(&thread1ID, NULL, thread1Name, NULL);   
    pthread_create(&thread2ID, NULL, thread2Name, NULL);   
    pthread_create(&thread3ID, NULL, thread3Name, NULL);

```
this may be analog to 
Scheduler.start(NULL, threadname);

if( Scheduler.running(threadname) ) would mean the thread has already been started, so that it is running or active 
- the same or sth similar like as an existing/true pthread thread ID != 0, e.g.
if (threadID) ...
for pthread, 

self-terminating would mean sth like
 if (expression) return;
inside a thread ( void  * functionname ) for pthread:
```
void* thread2Name(void *) {
  while( true ) { 
    ///... (code/loop)   
    delay(10);
    if (expression) return;
  }
  return NULL;
}
```
this might be mimicked by a
Scheduler.return();

terminating from another function (or thread) would mean sth like
`int pthread_cancel(pthread_t threadID);  `
for pthread,
so mimicked e.f. by a
Scheduler.cancel(threadname) or Scheduler.stopLoop(threadname)

for the location inside the thread to be canceled one may have  sort of "breakpoint", placed in the thread function, similar like one would do for the time-slice  scheduling  command yield() (breakpoint to switch to next time slice):
`pthread_testcancel()`
that might be mimicked by a
Scheduler.testcancel() or Scheduler.breakpoint

I hope I could make that clear, just having written that "impromptu" without re-checking the pthread libs in detail.

mikaelpatel, 2017-01-26:
@shiftleftplusone The Arduino Scheduler API is much simpler than POSIX Threads. There a lot missing.

The most important is the thread identity parameter. That is the handle for most functions. There is no handle in the Arduino Scheduler API. Using the function as handle can only work if we restrict tasks to always have a unique loop-function. An alternative is to change the return value from start/startLoop from bool to int and return a task handle. This could then be used in the rest of the extended API. 

The next issue is the semantics of "cancel" (or "stopLoop"). As context switching only occurs on yield the canceled task has to detect the "cancel" and perform necessary house keeping. 

Last, terminated tasks become the responsibility of the sketch as the library does not keep track of this. Task are assumed to be static. This implies that there is a need for "restart" function. 

All this adds a lot of complexity for functionality that I guess very few will ever use. I am somewhat amused by your proposal as you started by criticizing the API as too complex as there is an extra parameter function parameter (setup) to "start".

Cheers!

shiftleftplusone, 2017-01-26:
no, it was not too complex, but the examples appeared very weird to me  and they  did explain it not clearly enough in simple words how to start and what one had to do to start multiple threads - in the scheduler by C. Maglie there was just 1 example, but that one was so simple that it was self-explaining.
Anyway, I am not a C++ programmer but just ANSI C (C99) and so I don't understand much about  objects, and it was also an idea because just to have perpetual loops is very restricting   IMO.
In fact that was the most significant point  reason why I changed from my Due (feat. Scheduler) to the Pi (feat. pthread). Here I was able to do things where Scheduler failed completely (e.g., for PID control).
Perhaps you might implement a semaphore for each "loop" which will be set to "true" when the "loop" once has been started and then back to "false" when it has to be terminated? or perhaps a "loop-watcher" by a private background task? Or perhaps you implement your own list of "loop-IDs"  which mimic pthread threadID2?
No idea.
For simple projects it's fine to have your scheduler working both for ARMs and AVRs, that really good, my idea was just an idea to improve and enhance things also for more powerful MCUs (i.e. now for the Due, and for the future also for the Star Otto, wishfully).
Or one would have to implement a completely new pre-emptive multithreading lib with simple Arduino API  wraps  from the scratch, which provide most of the things featured by POSIX pthread and or C++11 threads.
Instead, one would have only the choice to migrate to SoCs like the Pi, for example.

mikaelpatel, 2017-01-26:
@shiftleftplusone Actually I think I would be easier to address this issue by implementing a sub-set of the POSIX Threads API for the Arduino.

shiftleftplusone, 2017-01-26:
well, by which way ever - I think it would be a good choice to have a way a Scheduler task to be stopped and optionally  re-started.

closed 2017-02-06 by mikaelpatel