mikaelpatel/Arduino-Scheduler #18: issue with this Scheduler version vs. cmaglie's Scheduler

shiftleftplusone, 2017-02-14:
hi,
I just run into this issue with my Due:

the former code using cmaglie's Scheduler workes and still works fine,
but with this (mpatel's) Scheduler the program freezes after
 curlf();   sprintf(sbuf, " ...", _curx_); lcdprint(sbuf);   delay(333);   // <<<<<<<<<<<<<< 

(last line in setup() ! )

what might be the reason?

```

/*   Arduino Remote Console     
 *   UART / Bluetooth HC-05 (master)
 *   Arduino Due
 *   IDE 1.6.5
 *    
 *   version RCsender0013
 *  
 * change log: 
 *   "heartbeat"
 *   
 */

 
#include <SPI.h>
#include <Scheduler.h>

// SPI header            //             |--|
// SPI pins Due          //  -----------|  |------------
  #define DUEMISO     74 //  |  RES   76_SCK   74_MISO |
  #define DUEMOSI     75 //  | -GND   75_MOSI   Vc+5V  |
  #define DUESCK      76 //  --------------------------

// SPI pins Mega
//#define MISO     50
//#define MOSI     51
//#define SCK      52

//=====================================================================================
// TFT LCD
//=====================================================================================

#define   UTFT_SmallFont     8 // UTFT 8x10
#define   UTFT_MediumFont   12 // UTFT ++
#define   UTFT_BigFont      18 // UTFT +++
#define   _SmallFont_        1 // 9341 6x9
#define   _MediumFont_       2 // 9341 12x16
#define   _BigFont_          3 // 9341 18x23

int16_t  LCDmaxX , LCDmaxY ;                // display size
int16_t  _curx_, _cury_,                    // last x,y cursor pos on TFT screen
         _maxx_, _maxy_;                    // max. x,y cursor pos on TFT screen       

// set LCD TFT type
int16_t  LCDTYPE   =   -1;

#define  _LCD1602_    1  // LCD1602  Hitachi HD44780 driver <LiquidCrystal.h>
                           // http://www.arduino.cc/en/Tutorial/LiquidCrystal   //
#define  _UTFT_       4  // Henning Karlsen UTFT 2.2-2.4" 220x176 - 320x240 lib
                           // http://henningkarlsen.com/electronics/library.php?id=51   //
#define _ILI9341_     8  // https://github.com/adafruit/Adafruit_ILI9340
                           // https://github.com/adafruit/Adafruit-GFX-Library //
#define _ILI9341due_  9  // ILI9341_due NEW lib by Marek Buriak
                           // http://marekburiak.github.io/ILI9341_due/ //
                           
//--------------------------------------------------------------------------------------------------

#define   tft_rst      0 
[... 661 lines cut ...]
   curlf(); 
   lcdprint(sbuf);
   Scheduler.startLoop(loop3); 
   sprintf(sbuf, "MT heartbeat task started "); 
   curlf(); 
   lcdprint(sbuf);
   curlf(); 

   sprintf(sbuf, "setup(): done.");
   Serial.println(); Serial.println(sbuf);   
   curlf(); curlf(); lcdprint(sbuf);

   sprintf(sbuf, "press JoystickButton 0 to start!");
   Serial.println(); Serial.println(sbuf);   
   curlf(); curlf(); lcdprint(sbuf);

   while( !sensortouch( joystBtn_0 ) );   while( sensortouch( joystBtn_0 ) );
   lcdcls(); 
   curlf();   sprintf(sbuf, " ...", _curx_); lcdprint(sbuf);   delay(333);   // <<<<<<<<<<<<<< 
     
   lcdcls(); 

}

```

mikaelpatel, 2017-02-14:
I would check the stack size. This sketch uses a lot of stack.

closed 2017-02-14 by mikaelpatel

shiftleftplusone, 2017-02-14:
well perhaps, but all required stack should be managed by the lib automatically (after all it's a Due!), and with cmaglie's lib it works nevertheless.

mikaelpatel, 2017-02-14:
The only possible action to avoid this is using the same default stack size. Actually I would like to go the other direction and force explicit stack size.

shiftleftplusone, 2017-02-14:
for using multithreding I actually wouldn't like to have to calculate all that by myself for either thread manually, but maybe your lib may give a warning during runtime if the stack is running too low?

mikaelpatel, 2017-02-14:
The arduino-library scheduler default stack is 1024 https://github.com/arduino-libraries/Scheduler/blob/master/src/Scheduler.h. This library uses a default value of 512. 

There is a support function stack() that the running task can use to assert the available headroom.

mikaelpatel, 2017-02-14:
POSIX threads on Linux use Mbytes of thread stack. Embedded requires higher knowledge and skills.

shiftleftplusone, 2017-02-14:
so how to resove this problem  for my encountered issue in detail?

mikaelpatel, 2017-02-14:
Start by changing the default stack size to 1024. After that use the stack() member function to determine how much stack is actually used. Trim the stack size accordingly but remember that the stack must always have room for any ISR. 

I should also state that I normally take a consultant fee :)

shiftleftplusone, 2017-02-14:
how can I change the default stack size to 1024? I never did things like that before :-/

I should also state that I normally take a debugging fee :)

mikaelpatel, 2017-02-14:
Have you check the stack size parameter?

shiftleftplusone, 2017-02-15:
I absolutely don't know how to do or to start to do anything to fix the problem because the program gets completely stuck and does nothing at all any more, and I do not understand anything of what you wrote about default stack or change default or check  stack. There is no comprehensible tutorial and the  given information is far too poor yet, unfortunately.
Using cmaglie's scheduler instead works out of the box, no weird adjustments needed.

mikaelpatel, 2017-02-15:
Start by explicitly adding the stack size to all task starts, e.g. Scheduler.startLoop(loop2, 1024). You might also need to use the Scheduler.begin(1024) to force the main task stack size. 

I would suggest that you stick to the original library if you are having problems. They have a support organization. I have a cat.

mikaelpatel, 2017-02-15:
The alternative is that you change the default values in the Scheduler header file, e.g. in https://github.com/mikaelpatel/Arduino-Scheduler/blob/master/Scheduler.h#L127 change 512 to 1024 and test.

mikaelpatel, 2017-02-15:
Please run the https://github.com/mikaelpatel/Arduino-Scheduler/blob/master/examples/SchedulerBlinkMax/SchedulerBlinkMax.ino example sketch to get an understanding for how the Scheduler.stack() member function works. It is very enlightening. 

Good luck with your project.

mikaelpatel, 2017-02-15:
Please see commit https://github.com/mikaelpatel/Arduino-Scheduler/commit/95a0f8100bbc8839fd7a19dae8b71d4c498950c9.

shiftleftplusone, 2017-02-15:
thank you, Now it's clear, I will do it this way:

> Start by explicitly adding the stack size to all task starts, e.g. Scheduler.startLoop(loop2, 1024). You might also need to use the Scheduler.begin(1024); to force the main task stack size.

changing in the header file is no option because of platform-compatibility reasons.

Unfortunately don't understand  the https://github.com/mikaelpatel/Arduino-Scheduler/blob/master/examples/SchedulerBlinkMax/SchedulerBlinkMax.ino example sketch at all, too many uncomprehensible letters... but it does not matter as far as Scheduler.startLoop(loop2, 1024) will work.

BTW, I have another DUE project  with 1 task requiring about  40kB temporary memory for large arrays of double plus 8 smaller ones with about 1-10kB each; in that case the task has to be started with 
Scheduler.startLoop(loop2, 40*1024) , and the other ones accordingly, correct?

shiftleftplusone, 2017-02-15:
also with 
Scheduler.startLoop(loop2, 1024) 
for all loops the programs freezes, nothing works.
Note that for loop() a 1024 cannot be specified.
I created a new loop1() for that as a substitute ans started it with 1024,
nothing changed.
The program does not run any more.

with cmaglie's Scheduler everything is fine though.

Try the code by yourself!

shiftleftplusone, 2017-02-15:
please reopen the issue as it is not resolved!

mikaelpatel, 2017-02-15:
Download the updated version with the same default stack as the arduino-libraries/Scheduler. Please note that it is not enough to add the stack size to startLoop calls. You also need to give the main task a larger stack.

shiftleftplusone, 2017-02-15:
I created a new loop1() for that as a substitute and started it with 1024,
nothing changed.
The program does not run any more and hangs up immediately after setup().

with cmaglie's Scheduler everything is fine though.

Try the code by yourself!

shiftleftplusone, 2017-02-15:
with the new downloaded version the program does not block after setup but then switches off completely a few seconds later, no screen output visible any more.

mikaelpatel, 2017-02-15:
@shiftleftplusone As I suspected the default stack size was too small. Now the next step is to adjust the main task stack size further. The arduino-libraries/Scheduler main task does not have a 1K default. It is limited dynamically by the head end. 

Please add the line `Scheduler.begin(2048);` in beginning of your setup(). The 2048 is just a number to start with as 1024 was too small. The main task, loop(), in your sketch uses a lot of stack. You can later use the `Scheduler.stack()` member function to determine the actual stack max size.

shiftleftplusone, 2017-02-15:
but with cmaglie's Scheduler everything is fine, so the main task stack size can't be the reason why.

shiftleftplusone, 2017-02-15:
Additionally, loop() does nothing any more.
All of the old loop is now in a new loop1().
Nevertheless, the issue remains the same.

mikaelpatel, 2017-02-15:
@shiftleftplusone Are you saying that loop() is empty???

shiftleftplusone, 2017-02-15:
yes.

mikaelpatel, 2017-02-15:
It has to contain a yield() or a delay().

shiftleftplusone, 2017-02-15:
```
/*   Arduino Remote Console     
 *   UART / Bluetooth HC-05 (master)
 *   Arduino Due
 *   IDE 1.6.12
 *    
 *   version RCsender0013
 *  
 * change log: 
 *   "heartbeat"
 *   
 */
 
// (C) Helmut Wunder (HaWe) 2015
// freie Verwendung für private Zwecke
// für kommerzielle Zwecke nur nach Genehmigung durch den Autor.
// Programming language: Arduino Sketch C/C++ (IDE 1.6.1 - 1.6.5)
// protected under the friendly Creative Commons Attribution-NonCommercial-ShareAlike 3.0 Unported License
// http://creativecommons.org/licenses/by-nc-sa/3.0/

 
#include <SPI.h>
#include <Scheduler.h>

// SPI header            //             |--|
// SPI pins Due          //  -----------|  |------------
  #define DUEMISO     74 //  |  RES   76_SCK   74_MISO |
  #define DUEMOSI     75 //  | -GND   75_MOSI   Vc+5V  |
  #define DUESCK      76 //  --------------------------

// SPI pins Mega
//#define MISO     50
//#define MOSI     51
//#define SCK      52

//=====================================================================================
// TFT LCD
//=====================================================================================

#define   UTFT_SmallFont     8 // UTFT 8x10
#define   UTFT_MediumFont   12 // UTFT ++
#define   UTFT_BigFont      18 // UTFT +++
#define   _SmallFont_        1 // 9341 6x9
#define   _MediumFont_       2 // 9341 12x16
#define   _BigFont_          3 // 9341 18x23

int16_t  LCDmaxX , LCDmaxY ;                // display size
int16_t  _curx_, _cury_,                    // last x,y cursor pos on TFT screen
         _maxx_, _maxy_;                    // max. x,y cursor pos on TFT screen       

// set LCD TFT type
int16_t  LCDTYPE   =   -1;

#define  _LCD1602_    1  // LCD1602  Hitachi HD44780 driver <LiquidCrystal.h>
                           // http://www.arduino.cc/en/Tutorial/LiquidCrystal   //
#define  _UTFT_       4  // Henning Karlsen UTFT 2.2-2.4" 220x176 - 320x240 lib
                           // http://henningkarlsen.com/electronics/library.php?id=51   //
#define _ILI9341_     8  // https://github.com/adafruit/Adafruit_ILI9340
                           // https://github.com/adafruit/Adafruit-GFX-Library //
#define _ILI9341due_  9  // ILI9341_due NEW lib by Marek Buriak
                           // http://marekburiak.github.io/ILI9341_due/ //
                           
//--------------------------------------------------------------------------------------------------

#define   tft_rst      0 
#define   tft_cs      48
#define   tft_dc      49
#define   _MEGAMISO_  50   
#define   _MEGAMOSI_  51
#define   _MEGASCK_   52
[... 664 lines cut ...]
   curlf(); 

   sprintf(sbuf, "setup(): done.");
   Serial.println(); Serial.println(sbuf);   
   curlf(); curlf(); lcdprint(sbuf);

   sprintf(sbuf, "press JoystickButton 0 to start!");
   Serial.println(); Serial.println(sbuf);   
   curlf(); curlf(); lcdprint(sbuf);

   while( !sensortouch( joystBtn_0 ) );   while( sensortouch( joystBtn_0 ) );
   lcdcls(); 
   curlf();   sprintf(sbuf, " ...", _curx_); lcdprint(sbuf);   delay(333);   // <<<<<<<<<<<<<< 
     
   lcdcls(); 

}

void loop() {
  
}

//=====================================================================================
//=====================================================================================
```

mikaelpatel, 2017-02-15:
So there is your error :)

shiftleftplusone, 2017-02-15:
no one told that before, and again:
cmaglie's version is fine for either loop or loop1

shiftleftplusone, 2017-02-15:
I think you must adjust all your defaults to make it work out of the box, like cmaglie did.

mikaelpatel, 2017-02-15:
Thanks for your suggestion.

shiftleftplusone, 2017-02-15:
also Scheduler. start does not work any more,
and also with yield() in loop() the program terminates prematurely.

So this issue is not resolved at all, no reason to close it.
Please reopen it again.

mikaelpatel, 2017-02-15:
I am sorry but I cannot help you with that. This issue is now locked.