mikaelpatel/Arduino-Scheduler #23: Extension to isolated non-looping tasks ?

rstoica, 2017-08-14:
Hi,

first of all thank you for releasing this really nice cooperative scheduler for Arduinos! I have been using it throughout my projects lately to get a more flexible and real-time feel from my Arduino related projects.

I opened this issue to ask whether it is planned in the roadmap of this project to support isolated non-looping threads (run to completion) ?

mikaelpatel, 2017-08-16:
Thanks for your interest in this Arduino library. 

I believe that "run-to-completion" tasks can simply be down with a task (function pointer/closure) queue and a set of worker threads (one or many). This is a very common design pattern. All the necessary functionality (and support) is already available. 

Obviously an explicit API extension could be added to the Scheduler API (class). 

I will see if I can make the time to add an example sketch for starters. The current [SchedulerEventQueue](https://github.com/mikaelpatel/Arduino-Scheduler/blob/master/examples/SchedulerEventQueue/SchedulerEventQueue.ino) example sketch could be used for this, i.e., changing the event queue from a queue of unsigned int, 

`Queue<unsigned int, 8> eventq;` 

to a queue of function pointers (func_t) 

`Queue<SchedulerClass::func_t, 8> eventq;`.

Cheers! Mikael

rstoica, 2017-08-17:
Hi Mikael,

Thanks for pointing out this functionality. I was doing something similar with std::deque, missed somehow the queue functionality and was looking to understand if you see valuable adding API support to the SchedulerClass itself.

Regarding an example sketch I put something quick together on my develop branch here https://github.com/rstoica/Arduino-Scheduler/commit/dba6a9525046be30740ae3f56ce3a34ea5aa64c9

Let me know if it is what you had in mind and I can open a PR if you want me to.

mikaelpatel, 2017-08-17:
Good Job! 

Open a PR and I will accept it. That was exactly what I was considering as an initial example. The next level would be to add an environment pointer to the task function, or a class with a virtual member function (e.g. run). 

Another possible extension is to show how to reschedule. 

Cheers! Mikael

rstoica, 2017-08-18:
Okay, opened PR https://github.com/mikaelpatel/Arduino-Scheduler/pull/24.

closed 2017-08-20 by mikaelpatel