mikaelpatel/Arduino-Scheduler #25: If the loop() function is empty, the other scheduled loop functions never execute.

gazialankus, 2017-11-12:
This library uses both the loop() and the other scheduled loop functions (loop1(), loop2(), etc.)

However, there is an issue: if the main loop() function is empty, loop1(), loop2() etc. never execute. 

This is clearly a bug.

mikaelpatel, 2017-11-12:
@gazialankus 
Thanks for your interest in this Arduino library. 

The answer to your question may be found on second line in the README; "The tasks are run until they call yield() or delay()". 

The loop() must contain yield() or delay() for a context switch to occur. The loop() is executed in a main task. 

Collaborative multi-tasking requires a task to yield control, there is no explicit context switch (for instance on timer interrupt). 

The main reason for not implementing implicit context switching is that the Arduino core and drivers will not work.

gazialankus, 2017-11-12:
Ah, that's right, my bad. However this is an easy trap to fall into. I think providing a more explicit warning in the README could help people like me, something like "Each loop function, including the default loop function, has to have a delay() or a yield() call, otherwise the others will never execute."

I really like the simplicity of this library over the alternatives, which makes it especially suitable for newbies.

closed 2017-11-14 by mikaelpatel