mikaelpatel/Arduino-Scheduler #26: Passing arguments to function

axelbusch, 2018-09-21:
I try to adjust the SchedulerTaskCompletionQueue example. I would like to pass some arguments to the function that is executed by the worker. Such as:

void print(char c){
  Serial.println(c);
}
[..]
char c = 'a';
task_t task = print(c);
taskq.push(&task);

Unfortunately I can only pass functions without any parameters. Is that possible and how?

mikaelpatel, 2018-09-21:
Nothing is impossible :) there are several approaches depending how high level syntax is wanted. The simplest is to add an environment pointer i.e. void* as a generic pointer to the task function.

axelbusch, 2018-09-23:
Great idea! Thank you, that worked for me!

mikaelpatel, 2018-09-23:
@axelbusch Great news! That is actually how it works in Cosa, one of my Arduino core projects; https://github.com/mikaelpatel/Cosa/blob/master/libraries/Nucleo/Thread.cpp#L72

closed 2018-09-24 by mikaelpatel