mikaelpatel/Arduino-Scheduler #28: Sleep is not working

tatulea, 2018-09-24:
Hi,

I have the main loop and another one with startLoop. From the main one, I am calling LowPower.sleep(), but it looks like both of the loops are still working and nothing happend.

mikaelpatel, 2018-09-24:
@tatulea What is the problem with the Scheduler? Did you forget to add a delay() or yield()?

tatulea, 2018-09-24:
I added yield, but is not working. Without having the second loop, it is working.

mikaelpatel, 2018-09-24:
@tatulea Please explain "not working". What is your configuration? Hardware/Software etc. Please add as much info as possible and try to explain what is the problem with the Scheduler. As you might understand I cannot help you with the LowPower library, and hardware that is not supported or tested. Sorry but your issue description is much to vague.

mikaelpatel, 2018-09-24:
@tatulea Might as well add that the Scheduler is not tested with the Low Power Library. And therefore must be considered not supported.

closed 2018-09-26 by mikaelpatel