mikaelpatel/Arduino-Scheduler #38: Scheduler doesn’t work

alantrundle, 2022-04-27:
Scheduler works until I add the Adafruit SSD13006 library, and then for some reason it stops working 

basically it never seems to trigger my setup it loop tasks.

mikaelpatel, 2022-06-07:
Are you asking for help debugging? Seems like you might need to adjust task stack sizes.

closed 2022-09-12 by mikaelpatel

alantrundle, 2023-03-31:
Hi Mikael,

I am trying to get an Arduino project to work, for which I am using your Scheduler, but for the life of me, I am unable to get it to work, and I am at a loss with regards to debugging it using your Examples, and info / examples obtained from Google.

In essence, my program sends JSON from a HM-10 using Sofware serial to my Arduino, and then it sends PWM duty to 4 LED's on my board, and also sends input status changes from the board to the HM10 using serialized JSON for which I can use Swift to decode and display input info in form of coloured button boxes etc.

The idea is that I can write a simple BLE program in Swift to set the brightness of my LED's, and update my program when someone changes the potentiometer, or presses a button on my board.
It's a start...

It might be that I am trying to do too much in one of the stacks, or both, but I cannot seem to work out how to set the stack size, and even what to set it to.

I am using a Nano.

I have included a pic of my board, and also the program.

I would be very grateful if you could have a look and advise if you get a free half hour sometime.

[cid:682ea71e-fece-43fa-97ae-d2c7d4ddd13d]

Thank you

Alan

mikaelpatel, 2023-03-31:
Hi Alan,Some quick advice and questions. Why are you using the Schedular? What do you hope to achieve with it? What is the memory footprint for the sketch? How much of the Nanos 1 kbyte RAM are you using? Have you calculated the max size of the stacks eg call frame and local variables plus max interrupt frames? How much RAM is used for global variables? With larger sketches more memory housekeeping is needed especially on these small MCUs? You do understand how Software Serial works? It blocks context switching?

alantrundle, 2023-03-31:
Hi Mikael,

I wish I could answer your questions, however I’m not an expert programmer, by any means.
The program works if I put the whole thing in a single loop(), but it takes a good 300ms or possibly slightly longer for my LED’s and buzzer to change state after sending a JSON string over Bluetooth to my HM-10, so it’s very “laggy”.

The main reason for using Scheduler was to get the JSON serialisation (sending to HM10), and JSON deserialisation (receiving from HM10) tasks to happen at the same time in parallel some what, rather than one of the tasks being held up by the code above.

I did manage somewhat of getting your Scheduler to process one of the 2 tasks, and then getting the default loop() to process the other task, but sadly it’s not taken out my “laggy” JSON decoder routine.

I fully appreciate that there is no such thing as parallel multi tasking in a real sense on the Nano as its a single core MCU, but was hoping to use it to speed things up by multi tasking code blocking routines as two separate tasks, but it’s still not worked as anticipated.

Thanks

Alan

mikaelpatel, 2023-03-31:
Hi Alan. Thanks for some details on your design thoughts and assumptions. I could give you the quick fix but what is the fun in that and miss out on learning something. Let us start with you reading the code for Software Serial and we discussing on possible concurrency ie reading and writing at the same time. Even with a yield() between each send and receive of characters can this be concurrent? Actually the HM-10 is doing some packaging for transmission. This knowledge can be used for concurrency. Also you should learn how to gather memory statistics from the compiler and calculate the memory footprint. Next step after that is to understand how the compiler uses memory and how you can write code that minimises memory usage. Otherwise you will never know if you sketch will actually work. You could also look into the cost of using JSON. How much memory and CPU cycles are needed for the serialisation and deserialisation?

mikaelpatel, 2023-03-31:
BW what is the length of the JSON message sent/revived by the Arduino? What is the baud rate and Bluetooth bit rate? This could explain the timing. At 9600 bps it takes over a ms per byte. For message sizes of 100-200 bytes the turn around time is at least the amount in ms.

alantrundle, 2023-03-31:
Hi Mikael,

Yes getting to grips with the Arduino is something I really want to do, however for the moment I’m just trying to design a simple Arduino program so enable me to really do some Swift UI 6 coding, using an Adafruit BLE library I found, so really this is more about playing with Swift UI rather than getting down to the nitty gritty with the Arduino at this time.

JSON wise, well my SwiftUI would sent the following to my HM10

{“RLED”:1023,”OLED”:1023,”GLED”:1023,”BLED”:1023, “BUZZER”:1}

And my board would send the following if anything state wise changes :

{“RLED”:1023,”OLED”:1023,”GLED”:1023,”BLED”:1023, “POT”:100, “BUT1”:1, “BUT2”:0, “BUT3”:0, “BUZZER”:1}

A quick fix if you have one with a explanation of what the fix does, and how it works would be fantastic.

I was trying to run the TX and RX independently to speed up the processing of each task, however it’s still taking 300ms when the HM10 has received the JSON to action that on the LED’s and buzzer.

State changes are sent within milliseconds after pressing a button, or twiddling the pot.

Thanks

Alan

alantrundle, 2023-03-31:
Hi Mikael,

Please see below for the response I sent earlier, but more importantly I have updated my code a bit more, and have included it for you to see.

It is working. Sometimes I got gobble de gook when sending data from my HM-10 to my Arduino, which it wasn't doing before. Looks like the Baud is going screwy, but might be something possibly to do with how I am using the Scheduler ?

hit and miss.... Could be noisy DC, and lack of capacitor filtering ?

Still getting 300ms delay to update my LED's and buzzer when the HM-10 has received my JSON string from my HM-10 terminal on my iPhone.

Regards

Alan

alantrundle, 2023-04-01:
Hi Mikael,

Thanks for the input & help, however I think I have solved my problem.

The Scheduler wasn't the right thing to use in this case, however moving a few things about, shortening my JSON strings, and shifting things around I think It's there!

 I have used it previously to independently control the diming of LED's, so I l know it's definitely useful.

I think introducing Scheduler topped my RAM allowance, and it was corrupting my Strings, rather than my problem being Baud rate, or noisy DC as previously suggested, and due to my 1K RAM, the Scheduler was defo bombing out my little Nano.

I had to introduce a 200ms delay when sending JSON from my Arduino to the HM-10 overwise it became swamped with data, which basically kept crashing out my Terminal, and this particular delay() isn't affecting my response time when sending JSON to my Arduino via HM-10 either which is totally awesome, and since introducing it, i'm not having to force quit my BLE terminal, ever better.

I will leave you with a brief Video of my project if interested ?, and see attached for my final revision, for which I will use to prototype my Swift UI program.

The 300ms delay has now gone away too, so my Arduino pretty much updates as soon as I send JSON to it as you can observe from the video.

ON BUZZ LEDS 10%
{"RL":10,"OL":10,"GL":10,"BL":10,"BZ":1}

ON BUZZ LEDS 5%
{"RL":50,"OL":50,"GL":50,"BL":50,"BZ":1}

LEDS ON 100% BUZZ OFF
{"RL":1023,"OL":1023,"GL":1023,"BL":1023,"BZ":0}

ALL OFF
{"RL":0,"OL":0,"GL":0,"BL":0,"BZ":0}

[cid:a9ec5bbb-1439-4c9b-ad2a-fbd3848ffee1]

Apple Store - Free BLE Terminal for MACOS

[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/video_16x16.png]IMG_0411.MOV<https://1drv.ms/v/s!Avoe3Ii-kfkKmm6nQCQBU3fKE6Rm>

I think it's now time to drag out the Swift UI BLE class, and jig it around 🙂

Regards

Alan

mikaelpatel, 2023-04-01:
Hi Alan. Good job. If you reduce the message sizes further and remove the JSON usage altogether the response time will be even better. You can also increase the baud rate which requires some tinkering. Even better is to replace Software Serial which is the real issue here. Last of all using the Scheduler is just so wrong in this case. Cheers, Mikael.