mikaelpatel/Arduino-Scheduler #39: thread safe sleep()

Sylensky, 2023-02-08:
Hi,

Is it possible to add a popular `sleep(ms)` function back to the scheduler? I am looking for a similiar approach to a `vTaskDelay(ms)` from FreeRTOS for example.

So far there is just a `yield()` which context-switches to the next task immediately.

mikaelpatel, 2023-02-15:
@Sylensky 
Thanks for the question. Actually the Arduino core function delay(ms) uses yield() and thus gives the coroutine/thread version of sleep().

https://github.com/arduino/ArduinoCore-avr/blob/42fa4a1ea1b1b11d1cc0a60298e529d37f9d14bd/cores/arduino/wiring.c#L106

https://github.com/mikaelpatel/Arduino-Scheduler/blob/8f339ae52fc2751f9e4d113e29849d4413226e05/examples/SchedulerDebounce/SchedulerDebounce.ino#L67

Cheers, Mikael

Sylensky, 2023-02-16:
Oh thanks, this just pointed me into the direction of a bug in my code.

closed 2023-02-16 by Sylensky